afollestad/material-camera #129: gradle build will not build

BRETT71, 2016-10-16:
## compile 'com.afollestad:material-camera:0.4.2'

I added this version of material-camera to my gradle and a can not sync 

I am getting this 

Failed to resolve: com.afollestad:easyvideoplayer:0.2.13

I am missing something?

mdolnik, 2016-10-16:
Short answer is this has been covered a few times
https://github.com/afollestad/material-camera/issues?utf8=%E2%9C%93&q=easyvideoplayer%20

Otherwise this has worked for me:

Ensure Main gradle file has:
`allprojects {
  repositories {
        jcenter()
        maven {
            url "https://jitpack.io"
        }
        maven {
            url "https://dl.bintray.com/drummer-aidan/maven"
        }
    }
}`

Ensure App gradle file has:
`repositories {
    jcenter()
    maven { url "https://dl.bintray.com/drummer-aidan/maven" }
}`
and `
dependencies {
    compile 'com.afollestad:material-camera:0.4.2'
}`

... Sorry the code formatting line breaks aren't working for me...

You could also try an mimic the source code's sample app to see how it's done there.
(Remember to replace `compile project(':library')` with `compile 'com.afollestad:material-camera:0.4.2'`)

Otherwise if this doesn't work for some reason @mowcixo figured out a workaround: https://github.com/afollestad/material-camera/issues/106#issuecomment-249573086

afollestad, 2016-10-17:
@mdolnik fixed your code formatting, you weren't using the correct method of doing so

closed 2016-10-17 by afollestad