afollestad/material-camera #131: If I take 2 photos preview shows the first photo after taking the second.

Peter-Barrett, 2016-10-18:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.1`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `Nexus`
 - Device Name: `Nexus 5`

Also, please wrap Java with correct syntax highlighting.

```java
new MaterialCamera(DetailerSettingsFragment.this)
                                .stillShot()
                                .start(CAMERA_RQ);
```
 
###### Reproduction Steps

1. Take a photo.
2. Use it or store it.
3. Take another photo

###### Expected Result
New photo is shown in preview

###### Actual Result
Preview shows the previously taken photo but passes the data for the new photo

afollestad, 2016-10-18:
You are not using the latest version of the library. It should work in the latest version.

If not, another release is coming soon.

afollestad, 2016-10-18:
Re-opening just in case

sugarmanz, 2016-10-18:
I was able to reproduce the issue. I'm trying to find the cause. It's a rather strange issue.

sugarmanz, 2016-10-19:
From what I can tell, the issue is resolved in the current revision 37fe300. I'm not sure what fixed it, but the issue goes away when updating from 0.4.2 to the current revision.

closed 2016-10-19 by sugarmanz

mowcixo, 2016-10-19:
Yes, that was fixed in 27a987d2d4b27b68706c63e101556f09d6e1ac4b but there was no tag for holding it, and it still being in master.

rinku9900, 2016-10-21:
is the 0.4.3 version published yet?