afollestad/material-camera #132: Recorded video's length is longer than actual recording length

kr1stjans, 2016-10-21:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.1`
 - Device OS version: `5.0`
 - Device Manufacturer: `LG`
 - Device Name: `Nexus 5 Emulator`

Also, please wrap Java with correct syntax highlighting.

```java
  File saveFolder = new File(Environment.getExternalStorageDirectory(), "SportStory");
                if (!saveFolder.exists() && !saveFolder.mkdirs())
                    throw new RuntimeException("Unable to create save directory, make sure WRITE_EXTERNAL_STORAGE permission is granted.");

                new MaterialCamera(getActivity())
                        .saveDir(saveFolder)
                        .showPortraitWarning(false)
                        .defaultToFrontFacing(false)
                        .start(CAMERA_RQ);
```
 
###### Reproduction Steps

1. Record a 5 second video.
2. Play recorded video.

###### Expected Result
5 second video.

###### Actual Result
55 second video.

To clarify: The recorded video is a lot longer than expected. The first 5 seconds of the video is normal and it is the actual recorded video, but the rest of the video is empty. It also seems the video is somehow corrupted. When I play it in VLC player on computer, it crashes when it passes first 5 seconds. Any ideas?

afollestad, 2016-11-07:
I've never experienced any issues close to this. Have you guys, @sugarmanz @tylerkrupicka?

sugarmanz, 2016-11-07:
No, I haven't. But, I'm not sure that I completely trust the emulator. @kr1stjans Are you able to reproduce the issue on an actual device?

afollestad, 2016-11-07:
Good point, I really don't think it's an issue.

closed 2016-11-07 by afollestad

tylerkrupicka, 2016-11-07:
Never seen it on my devices