afollestad/material-camera #134: Dynamic switching between still shot and video mode ?

shawaf, 2016-10-26:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.1`
 - Device OS version: `7.0`
 - Device Manufacturer: `Huawei`
 - Device Name: `Nexus 6P`

Is There a way to dynamic switching between still shot and video mode using the ui ? 

Thanks.

tylerkrupicka, 2016-10-27:
This is not currently implemented, but I agree that it would be a good feature to have on the roadmap.

idish, 2016-10-28:
I will be glad for such feature