afollestad/material-camera #135: show camera preview full screen

jamariya, 2016-10-27:
Hello There,

This is not an issues, I wanted to checking is that possible to have full screen preview for stillshot and video? I am trying to implement camera in full screen screen mode? Please let me know if that possible. Here is something I am trying to implement but I am not sure it's good idea to implement. Also I have minimum target as 21.

If I replace this:
// Danger, W.R.! Attempting to use too large a preview size could  exceed the camera
                // bus' bandwidth limitation, resulting in gorgeous previews but the storage of
                // garbage capture data.
                mPreviewSize = chooseOptimalSize(map.getOutputSizes(SurfaceTexture.class),
                        rotatedPreviewWidth, rotatedPreviewHeight, maxPreviewWidth,
                        maxPreviewHeight, largest);

With this:

mPreviewSize = new Size(maxPreviewWidth, maxPreviewHeight);

Is that good idea?

I have tried to implement it this way but i was showing some extra blank space on right side of camera? May be because of the AutoFitTextureView setAspectRatio not fitting well.

Any help will be useful. Thanks in advance.

jamariya, 2016-11-03:
Hello,

Is there any updates here?  I am still looking for some good solution to display the camera and video preview in full screen? Thanks.

MartinMobilize, 2017-05-11:
Hi, any updates?