afollestad/material-camera #137: Max number of cameras are open, close previous cameras first

toidv, 2016-10-31:
###### Include the following:
 - Material Camera version: `0.4.1`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `LG`
 - Device Name: `Nexus 5`

```java
MaterialCamera camera = new MaterialCamera(activity)
                    .showPortraitWarning(true)
                    .allowRetry(true)
                    .defaultToFrontFacing(true)
                    .countdownSeconds(30);

            if (isWriteExternalStoragePermission) {
                File saveDir = new File(Environment.getExternalStorageDirectory(), "Inploi");
                saveDir.mkdirs();
                camera.saveDir(saveDir);
            }
            camera.start(CAMERA_RQ);
```
###### Actual Result
```
0-31 15:43:03.705 20480-20480/com.quomodo.inploi W/System.err: java.lang.Exception: Max number of cameras are open, close previous cameras first.
10-31 15:43:03.705 20480-20480/com.quomodo.inploi W/System.err:     at com.afollestad.materialcamera.internal.Camera2Fragment$2.onError(Camera2Fragment.java:167)
10-31 15:43:03.717 20480-20480/com.quomodo.inploi W/System.err:     at android.hardware.camera2.impl.CameraDeviceImpl$8.run(CameraDeviceImpl.java:316)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at android.os.Looper.loop(Looper.java:148)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5417)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
10-31 15:43:03.718 20480-20480/com.quomodo.inploi W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```

afollestad, 2016-11-07:
I've never experienced this issue.

There may be a camera lock, which is only fixed by restarting your device.

afollestad, 2016-11-07:
Since I haven't ever had this problem on Nexus or Pixel devices, I will assume it was a one time thing.

closed 2016-11-07 by afollestad