afollestad/material-camera #139: Disable front-facing /back-facing button

pjebs, 2016-11-02:
I am using this library for taking photos.

Is it possible to disable either the front-facing or back-facing camera to force a particular camera.

tylerkrupicka, 2016-11-03:
As far as I can tell you cannot force front or back facing if both are available. The only control related to those would be which one is the default and what icons are used.

pjebs, 2016-11-04:
There is this option: `defaultToFrontFacing`.

I would like to use that option and also have another option which disables the button that allows the customer to change camera. That way they are 'forced' to front-facing camera.

It definitely is a common use case for chat/telephone based apps.

tylerkrupicka, 2016-11-07:
Hey @pjebs: Your explanation of the feature makes sense, and I don't think it would be particularly difficult to add to the API. However, would you mind explaining the chat/telephone app use case a bit more? The chat apps I use don't seem to force a particular orientation so I'm just trying to get a better feel for whether other people would be looking for this feature as well.

Thanks!

pjebs, 2016-11-07:
I'm actually an ios developer who has just started learning Android in an attempt to port my app.
In iOS, video-chat apps like Facetime and skype only activate the front-facing camera since it doesn't make sense to use the back camera when you are supposed to look at the screen to see the other person and the other person is meant to see you.

For my app, that is a definite requirement but I don't know enough about Android to fork this library and figure out how to make the modifications.

I'm pretty sure it'd be quite easy to implement and I hope I have convinced you of a usecase that justifies it.

pjebs, 2016-11-07:
Does this library work with API16. I noticed that `minSdkVersion` in gradle says `14`, but does that mean it just won't crash (and/or use a reduced feature set) or will it work as intended on API 16.

tylerkrupicka, 2016-11-07:
Hey @pjebs, does your app need to be streaming the video? If so this library might not be for you, as it returns the video after it is done recording. This library should work back to 14, however lower APIs are tested less frequently as the devices are harder to come by. I usually test the new versions on my Razr Maxx which is on API 17 and it usually works well. I can look in to adding this feature when i have some free time as a branch or part of a release -- but I'm a full time student so I can't guarantee a time frame.

pjebs, 2016-11-07:
I'm actually just using this library for snaps hitting their face to create profile pic. Thanks @tylerkrupicka

tylerkrupicka, 2016-11-07:
Ah. Sounds good. Actually, @sugarmanz was able to get it working earlier so he will be uploading a branch soon.

tylerkrupicka, 2016-11-07:
https://github.com/afollestad/material-camera/commit/ab60cbe1e86b74b8c5bee9368619bee820a5af1f

pjebs, 2016-11-07:
Will I need to make changes to gradle dependency to use new commit?

afollestad, 2016-11-07:
I will have to publish a release to jCenter before you can access it.

tylerkrupicka, 2016-11-07:
@pjebs yeah its not in gradle yet. We'll do some testing on our devices to make sure it doesn't break anything and then merge it in to master. In the meantime if you want to test it you can download the material camera repo, switch to the branch, and run the sample app. Someone will post here when we merge it in.

tylerkrupicka, 2016-11-10:
I've finished up testing on my devices:

| Device              | Manufacturer | OS  | Switch Lock|
| ------------------- |:------------:|:---:|:-----:|
| Nexus 6p            | Huewei       |7.0|   ✓  |
| Nexus 5             | LG           |6.0.1|   ✓  |
| Droid Razr Maxx     | Motorolla    |4.4.2|   ✓  
| One M8           | HTC         |5.0.5|   ✓  |

No real gotchas, didn't have any problems toggling it for each device. It's worth noting that the API is `allowChangeCamera(false);` not `allowCameraChange` as is stated in the commit. @sugarmanz we need to make sure this gets added to the README. 

Also, @pjebs turns out my Maxx is actually API 16 and material camera is working well.

pjebs, 2016-11-10:
Thanks guys for the amazing work!

Will trump winning have any impact on the performance of the library?

closed 2016-12-04 by pjebs