afollestad/material-camera #143: Camera view is not exited after pressing "use image"

jeiselvan, 2016-11-10:
- [Yes ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [Yes] I have verified that I am using the latest version of Material Camera.
- [Yes] I have given my issue a non-generic title.

**Include the following:**
 - Material Camera version: `0.4.4`
 - Device OS version: `6.0,4.4`
 - Device Manufacturer: `Samsung / Huawei / Sony Xperia`
 - Device Name: `Samsung Galaxy S7 Edge, Sony Experia E2003`
 
 **Reproduction Steps**
1. Make sure phone rotation is unlocked.
2. Press Take new picture in a conversation.
3. Put the phone into landscape mode.
4. Take a photo.
5. While keeping the phone in landscape mode, press Use image.
6. The camera view is still be active, even though the image is being uploaded in the background. 

**Expected Result**
The camera view is exited once the Use image button is pressed.

**Actual Result**
The camera view is still active.

tylerkrupicka, 2016-11-10:
hey @jeiselvan: would you mind posting the java code you are using to invoke material-camera? That way we have the best chance of reproducing this because I've never seen it before. Also, would you mind clarifying what you mean by uploaded? Do you mean the onActivityResult is being called? Thanks.

jeiselvan, 2016-11-15:
Hi,

The java code I used to invoke the material camera and the way I am capturing the result as follows 

------------------------------------------------------------------------
  MaterialCamera materialCamera = new MaterialCamera(this)
                    .saveDir(saveDir)
                    .showPortraitWarning(false)
                    .allowRetry(true)
                    .continueTimerInPlayback(false)
                    .countdownSeconds(30f)
                    .defaultToFrontFacing(false);

            materialCamera.stillShot(); // launches the Camera in stillshot mode

            materialCamera.start(CAMERA_RQ);

------------------------------------------------------------------------------

For your second part of the question, yes the onActivityResult is called when the camera view is still active. Below is the onActivityResult code I used in my program.

 if (requestCode == CAMERA_RQ) {
            if (resultCode == RESULT_OK) {

                if ( (intent.getStringExtra("type")).equals("video")) {

                    final File file = new File(data.getData().getPath());

                    Intent resultIntent = new Intent();
                    resultIntent.putExtra("file_path", file.getAbsolutePath());
                    resultIntent.putExtra("type", "video");
                    setResult(CAMERA_RQ, resultIntent);
                    finish();

                }

                if ((intent.getStringExtra("type")).equals("image") ) {

                    final File file = new File(data.getData().getPath());

                    Intent resultIntent = new Intent();
                    resultIntent.putExtra("file_path", file.getAbsolutePath());
                    resultIntent.putExtra("type", "image");
                    setResult(CAMERA_RQ, resultIntent);
                    finish();
                }

            }

killingdjef, 2017-01-15:
I'm having the same bug. To reproduce:

- Start your activity and keep in portrait view

-Use following method to start MaterialCamera
`    public void takePicture(){
        new MaterialCamera(this)
                .stillShot()
                .start(CAMERA_RQ);
    }`

-Rotate screen to landscape view; the MaterialCamera should rotate the cameraview for you.
-Take a picture
-Select "Use stillshot"
-I remain on the same page.

I have received the following exception:
`Handler (android.os.Handler) {7e88c7c} sending message to a Handler on a dead thread
                                                                           java.lang.IllegalStateException: Handler (android.os.Handler) {7e88c7c} sending message to a Handler on a dead thread
                                                                               at android.os.MessageQueue.enqueueMessage(MessageQueue.java:574)
                                                                               at android.os.Handler.enqueueMessage(Handler.java:631)
                                                                               at android.os.Handler.sendMessageAtTime(Handler.java:600)
                                                                               at android.os.Handler.sendMessageDelayed(Handler.java:570)
                                                                               at android.os.Handler.post(Handler.java:326)
                                                                               at android.hardware.camera2.dispatch.HandlerDispatcher.dispatch(HandlerDispatcher.java:61)
                                                                               at android.hardware.camera2.dispatch.MethodNameInvoker.invoke(MethodNameInvoker.java:88)
                                                                               at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.onClosed(CallbackProxies.java:177)
                                                                               at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:387)
                                                                               at android.hardware.camera2.impl.CameraCaptureSessionImpl.finalize(CameraCaptureSessionImpl.java:598)
                                                                               at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:216)
                                                                               at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:194)
                                                                               at java.lang.Thread.run(Thread.java:818)`

Possible cause: I believe the activity used by MaterialCamera is destroyed on rotation. When a picture is taken and "Use stillshot" has been tapped, MaterialCamera is trying to callback the original activity which has been destroyed on rotation causing this error.

killingdjef, 2017-01-15:
I can also confirm that the same issue happens when taking a video. Shooting media, rotating and then saving/selecting said media fails to produce an OnActivityResult call.