afollestad/material-camera #146: Custom camera icons not showing in actual color of image

franklinexpress, 2016-11-22:
- [ x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x ] I have verified that I am using the latest version of Material Camera.
- [ x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy S7 Edge`

Also, please wrap Java with correct syntax highlighting.

```java
MaterialCamera materialCamera = new MaterialCamera(this)
            .showPortraitWarning(false)
            .countdownSeconds(10f)
            .autoSubmit(false)
            .allowRetry(true)
            .retryExits(true)
            .iconRecord(R.drawable.recordbigwebstart)
            .iconStop(R.drawable.recordbigweb)
            .qualityProfile(MaterialCamera.QUALITY_720P)
            .primaryColor(getResources().getColor(R.color.colorPrimary))
            .defaultToFrontFacing(true);
    materialCamera.start(CAMERA_RQ);
```
 
###### Reproduction Steps
I was working at some point but suddenly it just changed.

1. Set custom icon with color, for ex. Red
2. Run and the color of the icon won't be like the image
3. 

###### Expected Result
Actual image saved from photoshop as png

###### Actual Result
image is all white. The images are for start recording and stop recording. Tried different sizes, all the same

tylerkrupicka, 2016-12-01:
Hey @franklinexpress: I can take a look at this sometime in the next couple weeks. If you wouldn't mind uploading an example icon so I don't have to make one for testing that would be helpful. Thanks!

franklinexpress, 2016-12-06:
[image]

franklinexpress, 2016-12-06:
[image]

franklinexpress, 2016-12-06:
I just don't have much time to get into it now, I moved on to create my own camera but I'm still curious to see the end result of this, thanks!

afollestad, 2016-12-06:
This is currently intended behavior, icons are tinted white or black depending on the "lightness" of your accent color. That's kind of part of Material philosophy (action buttons).

franklinexpress, 2016-12-06:
Right, but don't want to follow that "philosophy", I was wondering where I could make the changes, wen't through the code for bit but no avail

afollestad, 2016-12-06:
https://github.com/afollestad/material-camera/blob/cde4b0c3f3317445e6ebf3995022e7f417b833cf/library/src/main/java/com/afollestad/materialcamera/internal/BaseCameraFragment.java#L102

closed 2016-12-06 by afollestad