afollestad/material-camera #148: "Couldn't find any suitable preview size"

djani24, 2016-12-01:
Following are the logs:

Couldn't find any suitable preview size
Using resolution: 4128x3096
Orientations: Sensor = 90˚, Device = 0˚, Display = 90˚

Code used:

    new MaterialCamera(this)
                .allowChangeCamera(false)
                .stillShot() // launches the Camera in still shot mode
                .start(ValueConstants.REQUEST_CAMERA);

Device used: Samsung Galaxy s4  (GT-I9500)
Android Version: 5.0.1

tylerkrupicka, 2016-12-01:
Thanks for submitting this. I tested still shot on an S4 a couple of months ago and it was working, which version of Material Camera are you using?

djani24, 2016-12-02:
Thanks for the reply. I'm using the latest version ("0.4.4").

Waterbyte, 2016-12-02:
Same problem with me too. 

Logcat Details:
Couldn't find any suitable preview size.
Using resolution: 2880x1728
Orientations: Sensor = 90˚, Device = 0˚, Display = 90˚

Device Model: Micromax A106
Android Version:5.0

(Note: Even though Camera preview is distorted, the stillshots themselves  are ok in quality when viewed from gallery.)

fess89, 2016-12-09:
Same issue on Samsing SM-T520 tablet, Android 4.4.2.
Logcat details:
Couldn't find any suitable preview size
using resolution: 3264x2448
Orientations: Sensor = 0˚, Device = 0˚, Display = 0˚
mali_winsys: new_window_surface returns 0x3000

tylerkrupicka, 2016-12-20:
I no longer have access to my friend's S4, so this will be more difficult to debug. @sugarmanz and I can try to reproduce on the S3 and we can ask around and see if anyone else has an S4. Thanks for the information @fess89.

gabbimark, 2017-01-31:
same for Moto G3. Camera wont open due to this error.

fess89, 2017-02-06:
Seems that I can make it work on Samsung SM-T520 with API 19 by adding the following:
`.videoPreferredHeight(480)`
Larger heights, e.g. 720, do not work.