afollestad/material-camera #149: Take multiple still photos

ggua5470, 2016-12-01:
I hope that it can support taking multiple still photos without exiting and reentering the new Activity.
Or anyone give me some quick hints about where should I modify the code?

Thanks!

sugarmanz, 2016-12-01:
Hmm. This could be an interesting feature. I think the main issue would be
that it would potentially change the format of what's returned from the
activity, which could break people's code. I'll take a look and let you
know what I think. If you wanted to take a look, I would suggest looking in
places where data is returned to the calling activity.

ggua5470, 2016-12-01:
@sugarmanz Yep, there should be a change of the return data. As for **not** breaking exisiting code, it may still return the last photo taken as the normal data, and if the "multiple" mode is enabled, it will also return an array of Files (or something like that).

In this mode, the library's Activity also need to keep track of an array of File, and do not exit when a single photo is done. @sugarmanz do you know where should I look at? I am just a newbie here. I do not understand why there are 2 CaptureActivity there...

afollestad, 2016-12-10:
This isn't what I intended originally for this library to do. It's not meant to be an entire camera app, it's meant to be something you use in apps that need to take a single photo or video for submission elsewhere.

closed 2016-12-10 by afollestad

ggua5470, 2016-12-10:
So what's different between using this library and using an intent to
trigger the default camera app?

^_^

afollestad, 2016-12-10:
More granular control. This library was made specifically because a previous employer of mine needed things like recording timeouts, which the stock cameras don't provide.