afollestad/material-camera #152: Stillshot image quality not so good ("artifacts")

ThomasSHA, 2016-12-23:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy A3`

(Same for other devices e.g. OnePlus One)

How i started the capture process:
```java
new MaterialCamera(fragment)
                .saveDir(fragment.getActivity().getFilesDir())
                .stillShot()
                .retryExits(false)
                .labelConfirm(R.string.ok)
                .labelRetry(R.string.retry)
                .start(REQUEST_IMAGE_CAPTURE);
```
 
###### Reproduction Steps

1. Take a picture
2. look at preview
3. save the image

###### Expected Result
nice image, like with other camera-apps

###### Actual Result
image has red and blue "artifacts" all over it, especially visible on black surfaces (also in preview, before saving)

I uploaded a quick comparison: http://imgur.com/a/s77md (but had even worse results)

It seems like a "problem" with the post processing?!

ThomasSHA, 2017-01-10:
Just to make it clear this is NOT a Samsung only bug...

aftabsikander, 2017-01-26:
I am having the same issue. while taking picture camera surface is blurred alot, no matter how many times I manual auto focus preview stays blurred.

**Device Information**
- Material Camera version: 0.4.4
- Device OS version: 6.0.1
- Device Manufacturer: Samsung
- Device Name: S5
- Device Model: SM-G900P

**Expected Result**
Camera preview should not be blurred just like native camera application. 

**Actual Result**
Camera preview is blurred when using material camera. 

**Images comparison**

- [Device camera preview](https://cloud.githubusercontent.com/assets/4800511/22330546/937ea2f0-e3e8-11e6-863b-25a8f7ede2c7.jpg)
- [material camera preview](https://cloud.githubusercontent.com/assets/4800511/22330547/937ea91c-e3e8-11e6-90b1-8213c63f3722.png)

**Note:**
I believe this may be issue related to Camera API 2.

larrytech7, 2017-05-20:
This issue is a real one. I feel the images from material camera should be like native camera. this should be the goal if not better

akash-pal, 2017-07-25:
Is there any progress on this issue?