afollestad/material-camera #154: Customize video player layout ?

aftabsikander, 2016-12-27:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

Is there a simple way to customize the video preview layout when we play video after its taken. Currently the UI does not meet our app branding and we wanted to know if there is a way to customize the UI theme and other widgets for video player component.

afollestad, 2016-12-27:
Currently, no. The video player is from https://github.com/afollestad/easy-video-player

aftabsikander, 2016-12-28:
@afollestad Thanks for the link, I read the [Getting Started](https://github.com/afollestad/easy-video-player#getting-started) for easy-video-player, we can customize the appearance if we use that library individually.
But can't this be done using this library. I am sure you can link those attribute into this library to accommodate these layout customization.

BabuinGH, 2017-06-09:
@aftabsikander , I have a requirement to play the video center vertically with 1:1 ratio. Please, could you suggest me how to get this done using easy-video-player ?

meshileya, 2018-03-29:
@BabuinGH were you able to get it done???

BabuinGH, 2018-03-29:
@meshileya , I did get it working using FFMPEG Android http://writingminds.github.io/ffmpeg-android-java/
However, I'm now facing an issue on devices with Android 7. The cropping is failing on these devices.

meshileya, 2018-03-30:
That's serious, what's the error you were getting...moreover, what about the time lag...while using the FFMPEG library to crop the video???