afollestad/material-camera #155: Samsung S4 GT-i9500 can not open camera

yaochangliang159, 2017-01-09:
on samsung s4 gt-i9500 device, I click button "launch camera for recording",but it toasts that "error:cannot access the camera you may need to restart your devices",it also happens on other buttons

###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `4.4`
 - Device Manufacturer: `Samsung`
 - Device Name: `Glasxy S4`

aftabsikander, 2017-01-09:
@yaochangliang159  I assume this is happening because camera is being used by some other application and it forgot to handle release cases when application is not being caused,

Can you please confirm does you native camera application is opening prior to opening `material-camera`. I have the same device but I am unable to reproduce this.

halilozdemr, 2017-02-14:
on same device. 

yes i confirm this.
> Can you please confirm does you native camera application is opening prior to opening material-camera
 
[image]

tanmay-gp91, 2017-03-09:
Loved the library and almost pushed the code to production before it did not work on this Samsung device. 😐

pragneshghoda, 2017-03-22:
This same scenario happened with me in samsung s4 devices with same error. Library can not handle camera in samsung devices.

@afollestad  have you found any solution or atleast suggestion for this ?