afollestad/material-camera #167: Retrieving frame by frame

shaheenkdr, 2017-02-24:
Hello, trying to retrieve images frame by frame from the camera view finder. Does this library has any methods that helps in this regard? I tried the docs, however couldn't find any. Please help

tylerkrupicka, 2017-02-24:
What specifically are you trying to do? Are you hoping to get images from the screen instead of the native camera capture?

shaheenkdr, 2017-02-24:
I am building an image processing app, so I need to get frames from the camera view finder, so that I can pass them to the NDK to process the image . The typical `SurfaceView` has a similar option, however can't use that it has multiple issues with chinese phones like Mi, Oneplus etc . So I was looking if there's a reliable option available in this library. This is how I do in SurfaceView and Camera : 

```
private Camera.PreviewCallback previewCallBack = new Camera.PreviewCallback() {
    @Override
    public void onPreviewFrame(byte[] data, Camera camera) {

        System.out.println("Get frame " + frameNumber);
        if (data == null)
            throw new NullPointerException();
        Camera.Parameters p = camera.getParameters();
        Camera.Size size = p.getPreviewSize();
        if (frameNumber == 0) {
            startTime = System.currentTimeMillis();
        }
        // Log.e("GetData", "Get frame " + frameNumber);
        frameNumber++;

        camera.addCallbackBuffer(data);

            }
      }
```

shaheenkdr, 2017-02-24:
> What specifically are you trying to do? Are you hoping to get images from the screen instead of the native camera capture?

YES, to retrieve images from Screen :)

afollestad, 2017-02-28:
This is out of scope for this library

closed 2017-02-28 by afollestad