afollestad/material-camera #169: Null pointer exception when trying to lock camera

shalevmo, 2017-03-01:
###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `6..0.1`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy J7`

Fatal Exception: java.lang.NullPointerException
Attempt to invoke virtual method 'void android.hardware.Camera.lock()' on a null object reference
 Raw Text
com.afollestad.materialcamera.internal.CameraFragment.prepareMediaRecorder (CameraFragment.java:396)
com.afollestad.materialcamera.internal.CameraFragment.startRecordingVideo (CameraFragment.java:410)
com.afollestad.materialcamera.internal.BaseCameraFragment.onClick (BaseCameraFragment.java:410)
com.afollestad.materialcamera.internal.CameraFragment.onClick (CameraFragment.java:147)
 
###### Reproduction Steps
I don't know how to reproduce, I got this error on Crashlytics.
It also happen on Galaxy J5 (2016) running ANDROID 6.0.1 and Meizu m3 Note running ANDROID 5.1

geared554, 2017-04-21:
+1
ARK Benefit A2, 5.1.1, got issue report via Crashlytics

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.lock()' on a null object reference
       at com.afollestad.materialcamera.internal.CameraFragment.prepareMediaRecorder(CameraFragment.java:394)
       at com.afollestad.materialcamera.internal.CameraFragment.startRecordingVideo(CameraFragment.java:408)
       at com.afollestad.materialcamera.internal.BaseCameraFragment.onClick(BaseCameraFragment.java:409)
       at com.afollestad.materialcamera.internal.CameraFragment.onClick(CameraFragment.java:140)
       at android.view.View.performClick(View.java:4781)
       at android.view.View$PerformClick.run(View.java:19874)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.app.ActivityThread.main(ActivityThread.java:5254)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:902)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:697)
```

holakeshav, 2018-01-10:
The null check seems to missing.
Add this and try if it helps.

`if (mCamera != null) {
                    mCamera.lock();
                }`