afollestad/material-camera #170: Audio is played first before the video

Dyoed, 2017-03-02:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `7.0`
 - Device Manufacturer: `Samsung`
 - Device Name: `S6`

Video captured becomes 2x longer, the first half, the video and still and the audio is played. The second half, only the video is played
 
When I try recording with audio disabled, the problem doesn't occur.

Dyoed, 2017-03-06:
It seems like a device problem, I restarted the phone and it no longer occurs.

closed 2017-03-06 by Dyoed

wu16hk, 2017-03-16:
I have the same problem as you.