afollestad/material-camera #176: Incorrect Image Display After taking picture - Front Facing Camera - Still Shot - Rotated 180 degrees

ejezisek, 2017-04-05:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `7.1.1`
 - Device Name: `Pixel`

```java
// default camera orientation used to be 90 degrees, for Nexus 5X, 6P it is 270 degrees
if (sensorOrientation == Degrees.DEGREES_270) {
    displayRotation += 2 % 3;
}
```
should be:
```java
// default camera orientation used to be 90 degrees, for Nexus 5X, 6P it is 270 degrees
if (sensorOrientation == Degrees.DEGREES_270) {
    displayRotation = (displayRotation + 2) % 4;
}
 ```
The incorrect code above is equivalent to displayRotation += 2;  This needs to be updated to be either displayRotation+=2; displayRotation %= 4; or displayRotation=(displayRotation+2)%4;
There are essentially 4 different camera rotations, and as such %3 does not make sense. 
###### Reproduction Steps

1. Using the app with the front facing camera rotate the phone to the left. 
2. Take a picture.

###### Expected Result
I would expect to get a picture back that is rotated 0 degrees.

###### Actual Result
I get a picture back that is rotated 180 degrees.

vigneshpuc, 2017-11-21:
Still facing the same issue.

luminousman, 2017-12-27:
Still same issue is there