afollestad/material-camera #180: Video recording error on Android 7

sasa0, 2017-05-03:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.13` (same issue with the newest version)
 - Device OS version: `7.0`
 - Device Manufacturer: `Huawei`
 - Device Name: `Honor 7 Lite`

The camera worked fine on all devices I have checked up to Android Nougat. As soon as my system got updated to Android 7.0, the camera stopped working.
 
###### Reproduction Steps

1. Open the camera
2. The preview screen will freeze
3. Hit record button
4. After I hit stop recording, I get the following error message: https://drive.google.com/open?id=0BxxlgnWr4DnybkM4LW9qTzZiX28

redbluenat, 2017-11-28:
did you resolve your problem? @sasa0

sasa0, 2017-11-28:
No I didn't.

redbluenat, 2017-11-28:
I resolved this problem, I am using for android 7 CaptureActivity and then works everything

sasa0, 2017-11-29:
Have you reproduced the problem first? The issue does not exist on all devices running Android 7.