afollestad/material-camera #183: Can I save video file in internal storage

Dhrupal, 2017-05-12:
- [yes ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ Yes] I have verified that I am using the latest version of Material Camera.
- [ Yes] I have given my issue a non-generic title.

Can I save video file in internal storage?

lenoch0d, 2017-07-24:
Sure you can. 
```
final String myDir = "myDir";

File folder = new File(getActivity().getFilesDir(), myDir);
if (!folder.exists()) {
    boolean success = folder.mkdir();
    if (!success) {
        throw new IllegalStateException("Can't create dir.");
    }
}
new MaterialCamera(MyFragment.this)
          .saveDir(folder)
          .start(1001);
```

im4eversmrt, 2018-02-22:
how to disable save Dir ?