afollestad/material-camera #184: allowChangeCamera(false) its not working

jitarun, 2017-05-17:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.13`
 - Device OS version: `7.0`
 - Device Manufacturer: `Huawei`
 - Device Name: `Nexus 6P`

Also, please wrap Java with correct syntax highlighting.

```java
System.out.println("Hello, world!");
```
 
###### Reproduction Steps

1. 
2. 
3. 

###### Expected Result

###### Actual Result

rlee1990, 2017-08-10:
Any updates on this?

qadirsuh, 2017-08-29:
me too facing this.

OneXeor, 2018-03-05:
+