afollestad/material-camera #185: onActivityResult is not getting called on using material camera

ronneyismaelcodebakerz, 2017-05-19:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.13`
 - Device OS version: `7.0`
 - Device Manufacturer: `Huawei`
 - Device Name: `Nexus 6P`

Also, please wrap Java with correct syntax highlighting.

```java
System.out.println("Hello, world!");
```
 
###### Reproduction Steps

1. 
2. 
3. 

###### Expected Result

###### Actual Result

ronneyismaelcodebakerz, 2017-05-19:
When I click camera floating action button, I am able to take picture and it gets save on internal memory but it is not going to onActivityResult() method.

ronneyismaelcodebakerz, 2017-05-19:
# **This is my Code**

public class DeliveryDetailsFragment extends BaseFragment implements DeliveryDetailsMvp,
        View.OnClickListener, CompoundButton.OnCheckedChangeListener {
    // TODO: Rename parameter arguments, choose names that match
    // the fragment initialization parameters, e.g. ARG_ITEM_NUMBER
    private View rootView;
    private IndentModel indent;
    File saveDir = null;
    private ProgressDialog mProgressDialog;
    String filename;
    private final static int CAMERA_RQ = 7000;
    private final static int PERMISSION_RQ = 86;
    private DeliveryModel deliveryModel;
    private boolean isCheckListExpanded = false;
    private List<String> photoList = new ArrayList<String>();
    @BindView(R.id.tv_slno)
    TextView tvSlNo;
    @BindView(R.id.tvindentid)
    TextView tvIndentId;
    @BindView(R.id.tvstatus)
    TextView tvStatus;
    @BindView(R.id.recyclerview1)
    RecyclerView recyclerViewCheckbox;
    @BindView(R.id.recyclerview2)
    RecyclerView recyclerViewCamera;
    @BindView(R.id.button_delivery_accepted)
    SwipeButton mSwipeButton;
    @BindView(R.id.tv_materiialid)
    TextView materialId;
    @BindView(R.id.tv_checklistid)
    TextView checkListId;
    @BindView(R.id.android_floating_action_button_fab)
    FloatingActionButton fabCamera;
    @BindView(R.id.toggle_check_list)
    IconTextView togglechecklist;
    @BindView(R.id.checkBox_all)
    CheckBox checkboxAll;
    private static int count = 0;

    @BindView(R.id.checkBox_unselect)
    CheckBox checkBoxunselect;
    private boolean ischecked = false;

    @Inject
    DeliveryDetailsPresenter deliveryDetailsPresenter;
    @Inject
    DeliveryItemRecyclerView deliveryItemRecyclerView;

    private DeliveryCameraItemRecyclerView deliveryCameraItemRecyclerView;

    @Inject
    public DeliveryDetailsFragment() {
    }

    @Override
    public void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        fragmentComponent().inject(this);
        setRetainInstance(true);
        deliveryDetailsPresenter.attachView(this);

    }

    @Override
    public void onViewCreated(View view, @Nullable Bundle savedInstanceState) {
        super.onViewCreated(view, savedInstanceState);
        bindViews(view);
    }

[... 177 lines cut ...]
        } else {
            cursor.moveToFirst();
            int idx = cursor.getColumnIndex(MediaStore.Images.ImageColumns.DATA);
            return cursor.getString(idx);
        }
    }

    @Override
    public void onCheckedChanged(CompoundButton buttonView, boolean isChecked) {
        switch (buttonView.getId()) {
            case R.id.checkBox_all:
                deliveryItemRecyclerView.checkItemList(true);
                deliveryItemRecyclerView.notifyDataSetChanged();
                break;

            case R.id.checkBox_unselect:

                deliveryItemRecyclerView.checkItemList(false);

                deliveryItemRecyclerView.notifyDataSetChanged();
                break;
        }
    }

}

larrytech7, 2017-05-20:
I think i'm having the same issue. Needs to be addressed please

larrytech7, 2017-05-20:
I think i may have figured out the issue with the Activity callback not delivering the result in the onActivityResult method. You are using a fragment but when you initialize the material camera, you are passing the context of your calling activity and not the fragment's context, so that's why when the results come back, your fragment's onActivityResult callback is not triggered but rather the onActivityResult method of your host Activity will be called. What you need to use is the following;
`new MaterialCamera(this)`
instead of 
`new MaterialCamera(getActivity())`

This way, the result gets delivered first to your fragment before your host Activity. Fragments share a context with their activity but have their own life cycles so you need to explicitly define the context for many of such callbacks. Hope this helps someone.