afollestad/material-camera #193: Cannot access the camera, you may need to restart your device.

saefbadwan, 2017-06-28:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `4`
 - Device OS version: `4.4`
 - Device Manufacturer: `Samsuing`
 - Device Name: `Note 4`

- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `4`
 - Device OS version: `4.4`
 - Device Manufacturer: `Samsuing`
 - Device Name: `Note 4`

```
/CameraFragment: Couldn't find any suitable preview size
06-28 15:55:00.665 22079-22079/com.network.imyth.network D/CameraFragment: Using resolution: 4128x3096
06-28 15:55:00.665 22079-22079/com.network.imyth.network D/CameraFragment: Orientations: Sensor = ٩٠˚, Device = ٠˚, Display = ٩٠˚
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err: java.lang.Exception: Cannot access the camera, you may need to restart your device.
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at com.afollestad.materialcamera.internal.CameraFragment.openCamera(CameraFragment.java:254)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at com.afollestad.materialcamera.internal.CameraFragment.onResume(CameraFragment.java:119)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.Fragment.performResume(Fragment.java:2162)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:943)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1082)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1064)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.FragmentManagerImpl.dispatchResume(FragmentManager.java:1894)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.Activity.performResume(Activity.java:6421)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3383)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3425)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2763)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.ActivityThread.access$900(ActivityThread.java:177)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1448)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:102)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.os.Looper.loop(Looper.java:145)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5942)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at java.lang.reflect.Method.invoke(Method.java:372)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1400)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1195)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err: Caused by: java.lang.RuntimeException: setParameters failed
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.hardware.Camera.native_setParameters(Native Method)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at android.hardware.Camera.setParameters(Camera.java:1941)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err:     at com.afollestad.materialcamera.internal.CameraFragment.openCamera(CameraFragment.java:239)
06-28 15:55:00.820 22079-22079/com.network.imyth.network W/System.err: 	... 19 more
```

anshudwibhashi, 2017-06-29:
Hey. I seemed to have found why this happens. This is a common bug in Samsung devices. Samsung **requires** the `EXTRA_OUTPUT` extra for camera intents. Although you may've mentioned the directory to save the result in, through the `MaterialCamera.saveDir()` function, I just noticed that the `CameraFragment` class ignores the extra. I suspect this is why the extra isn't passed into the final camera intent, and Samsung devices crash. 

I too have _just_ discovered this bug, and am still investigating it. For now I'm going to add a function to the `CameraFragment` class to not ignore the extra.

P.S. I'm new to Github. If I do fix this bug, how do I contribute the new code to the original library?

searchy2, 2017-06-29:
Click the fork button and save this repo under your name. Then, pull the forked repo into your development environment. Make the changes necessary, then commit and push back to your repo. Finally, submit a PR. 

I would suggest using Gitkraken to manage your Git. http://gitkraken.com/
This article has pictures on submitting a PR: https://akrabat.com/the-beginners-guide-to-contributing-to-a-github-project/

anshudwibhashi, 2017-06-29:
Okay, I just discovered that I was wrong in my previous comment. MaterialCamera _does_ recognise the SAVE_DIR extra. Maybe it's that the CameraFragment class checks only for Samsung S3 and not all Samsung Devices?

Tgo1014, 2017-08-03:
I've the same problem in my Xperia

roiavr, 2018-01-07:
Someone solve this bug? it's happen to me on samsung Galaxy s4.