afollestad/material-camera #194: Error:Failed to resolve: com.afollestad:easyvideoplayer:0.3.0

13120241790, 2017-07-05:
Error:Failed to resolve: com.afollestad:easyvideoplayer:0.3.0
Hi ， need help~  This is why?

anshudwibhashi, 2017-07-05:
Add this to your gradle file:

`compile 'com.afollestad:easyvideoplayer:0.3.0'`

niravmandani, 2017-07-18:
Error:Failed to resolve: com.afollestad:easyvideoplayer:0.3.0 same issue please help

muhcobanoglu, 2017-07-21:
I have error same as @niravmandani. Please help us.

vinny1575, 2017-07-21:
Add this under Repositories in your gradle.
   
 maven { url 'https://dl.bintray.com/drummer-aidan/maven' }

andungu, 2017-07-27:
still failing even with maven { url 'https://dl.bintray.com/drummer-aidan/maven' }

mbronzovic-shiftgig, 2017-07-27:
Did someone resolve this? @afollestad in the sample you add you are using a the local reference of the library project. Do you know how to solve this issue? Do we need to fork the project?
Thanks

samlee99, 2017-07-31:
I got it to work by putting this into the project level build.gradle file (**NOT** the app level build.gradle):

buildscript {
    repositories {
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.3.3'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
        classpath 'com.google.gms:google-services:3.0.0'
    }
}

allprojects {
    repositories {
        jcenter()
        **maven { url "https://dl.bintray.com/drummer-aidan/maven" }**
    }
}

amirhosenebrahimi, 2018-03-18:
hi . i have problem to compile com.afollestad:easyvideoplayer:0.3.0 
I tried all the above solutions
But I did not get the result
pleas help me

bakyaraj008, 2018-03-19:
I can't able to stream the url --> "rtsp://admin:admin@192.XXX.X.XXX:554/12". Can anyone tell how to solve this. Waiting for your response.