afollestad/material-camera #197: Crash when starting MaterialCamera - LinkageError - Possible MultiDex Related

eygraber, 2017-07-19:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4`
 - Device OS version: `7.1.2`
 - Device Manufacturer: `Google`
 - Device Name: `Pixel`

Also, please wrap Java with correct syntax highlighting.
 
###### Reproduction Steps

1. Call `MaterialCamera.start()`

###### Expected Result

The camera should open

###### Actual Result

Crash with 

> java.lang.LinkageError: Method void com.afollestad.materialcamera.CaptureActivity2.onCreate(android.os.Bundle) overrides final method in class Lcom/afollestad/materialcamera/internal/BaseCaptureActivity; (declaration of 'com.afollestad.materialcamera.CaptureActivity2' appears in /data/app/com.myapp-1/split_lib_dependencies_apk.apk:classes6.dex)

ashik619, 2018-01-04:
@eygraber are you able to fix for this issue

meshileya, 2018-03-21:
@ashik619 @eygraber  how did you resolve this issue, I encountered same issue when using this library

meshileya, 2018-03-21:
api('com.afollestad:material-camera:0.4.4') {
        exclude module: 'support-v7'
        exclude module: 'support-v4'
        exclude module: 'support-v13'
    }

That solved my issue.