afollestad/material-camera #201: Camera Preview looking stretched

tigerbally, 2017-08-09:
Hi,

I am using material-camera for video recording but its recording preview looking stretched so Please provide a solution steps for resolve this issues means: "How can we maintain camera preview screen aspect ratio as per screen width and height?".

Waiting for your awesome reply...

"Thanking You"

Regards,
Bally Sharma

tigerbally, 2017-08-10:
Please reply me How can I resolve these issues?...

tigerbally, 2017-08-14:
How to resolve below issue:

Caused by: java.lang.
ClassNotFoundException: Didn't find class "me.zhanghai.android.materialprogressbar.MaterialProgressBar" on path: 
DexPathList[[zip file "/data/app/com.action60-1.apk"],nativeLibraryDirectories=[/data/app-lib/com.action60-1, /vendor/lib, /system/lib, /system/lib/arm]]

Awaiting for reply...

Tgo1014, 2017-08-14:
I'm too having the stretched problem. Although the final image is correct, the preview is distorted.

tigerbally, 2017-08-19:
Hi,

Crash issues I have fixed but camera preview stretched or distorted...if any solution let be know...we are trying to resolve this issues...

vishalpatel1327, 2017-09-14:
@tigerbally   can u tell me how to resolve stretching ?

aviontechnology, 2017-10-26:
I am too having camera preview stretched and blurry issue.
how to solve this issue.

tigerbally, 2017-10-30:
I m resolving these issues by changes in Library app and then looking exact camera preview screen not stretched or distorted ...

aviontechnology, 2017-10-30:
Thank you tigerbally,

Up to when you update library.

tigerbally, 2017-11-01:
Still me working on some more issues soon will be update...

aviontechnology, 2017-11-06:
Ok,
We guys waiting for this update.

aviontechnology, 2017-11-09:
Hello tigerbally, 

Can you please launch update with camera preview issue solution.
So i can use it in my application.if possible please launch it that way i can achieve my deadlines.
Thank you in advance.

aviontechnology, 2017-11-16:
Hello tigerbally,

Can you please reply upto when your publish new version.
So accordingly i can possible to do my next plans.
Please reply.

2020gaurav, 2017-11-18:
Can some one suggest any work around here ? We have been using material camera in our project, we are on edge of making our app live and now this issue is discovered.  @tigerbally You can share your findings here what is the cause or at least the progress. Might be you can have some help from the community. Please give us some temp fix if possible.

Thanks in advance.

aviontechnology, 2017-12-05:
Hello, 
Still waiting any work around for camera preview issue.
Please suggest if any one found any solution for this issue.

waiting ..

muhammadsr, 2018-03-02:
Having same issue here