afollestad/material-camera #204: Android OS 7 Two Crashes

sallison117, 2017-08-17:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4`

I got these crashes from users of my app. Reported to me through Crashlytics.
Below is the usage that I am using on a Fragment (android.support.v4.app.Fragment).
This crashes the app when they happen. There has been 20 counts between the two crashes.

Crashes 
StillshotPreviewFragment.jave line 90
com.afollestad.materialcamera.internal.StillshotPreviewFragment.setImageBitmap
Fatel Exception: java.lang.IllegalStateException
Fragment StillshotPreviewFragment{619e656} not attached to Activity
Android OS 7 (LGE: LGMS210 Mostly)

Camera2Fragment.java line 506
com.afollestad.materialcamera.interal.Camera2Fragment.openCamera
Fatel Exception: java.lang.ClassCastException
java.lang.Integer cannot be cast to java.lang.String
Android OS 7 (LGE: VS995 Multiple Crashes)

```java
new MaterialCamera(this)
				.showPortraitWarning(true)
				.qualityProfile(MaterialCamera.QUALITY_LOW)
				.maxAllowedFileSize(1024 * 1024 * 5)
				.stillShot()
				.start(CAPTURE_IMAGE_ACTIVITY_REQUEST_CODE);
```

android-bash, 2018-02-16:
Even i am facing the same issue some times in my Nokia 6 device. Did you find a solution??

sallison117, 2018-02-16:
I did not find a solution for these issues.