afollestad/material-camera #225: Error:(46, 0) Could not read script

Capotasto, 2018-01-08:
I have got grade error this morning when I open Android Studio.

- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.4.`
 - Device OS version: -
 - Device Manufacturer:  -
 - Device Name: `Nexus -

Also, please wrap Java with correct syntax highlighting.

```gradle
Error:(46, 0) Could not read script 'https://raw.githubusercontent.com/afollestad/aidanfollestad.com/master/android-lib-release.gradle' as it does not exist.
<a href="openFile:C:\Users\canoegi\develop\android\WearablesAndroid\library\build.gradle">Open File</a>
```

###### Reproduction Steps

1.Open the Android Studio 

###### Expected Result
Gradle project sync succeeded.

###### Actual Result
Gradle project sync failed.

Capotasto, 2018-01-09:
@afollestad 
I have just deleted one line below from `build.gradle` and fortunately, it's working but is this proper way to fix this problem?

```
apply from: 'https://raw.githubusercontent.com/afollestad/aidanfollestad.com/master/android-lib-release.gradle'
```

infineet, 2018-04-04:
I deleted the line,but still showing the same error.